robosoft-ai/SMACC #12: compile error while calling function

gisen-robo-git, 2019-11-12:
in tutorial（https://github.com/reelrbtx/SMACC#creatingaccessing-to-smacccomponents）
compile error in line `this->requiresComponent(moveBaseClient_);` like below
```
error: no matching function for call to ‘smacc_tutorial::BaseStateMachine::requiresComponent(smacc::SmaccMoveBaseActionClient*&, ros::NodeHandle&, std::__cxx11::string&, bool&)’
     base_type::outermost_context().requiresComponent(storage, nh, value, verbose);

...

error: no matching function for call to ‘processTransition(boost::statechart::transition<smacc::EvActionResult<smacc::SmaccMoveBaseActionClient>, smacc_tutorial::State2>*&, std::shared_ptr<smacc::SmaccStateInfo>&)’
     processTransition(dummy, sourceState);
```
could you please fix this bag

pabloinigoblasco, 2019-11-12:
Could you please copy the complete code you used?
We are currently in an important refactoring and some code snippets you see in the readme.md file may be outdated.

gisen-robo-git, 2019-11-19:
Thanks for your reply.
I'm sorry about my explanation is not enough.
Here is the complete code.

```
#include <ros/ros.h>
#include <smacc/smacc.h>
#include <smacc/smacc_state.h>
#include <smacc_navigation_plugin/move_base_to_goal.h>
#include <smacc/interface_components/smacc_topic_subscriber.h>
#include <smacc/interface_components/smacc_action_client_base.h>
#include <smacc/interface_components/smacc_service_client.h>
#include <std_msgs/UInt16.h>
#include <std_srvs/SetBool.h>
#include <tf/transform_datatypes.h>

namespace smacc_tutorial{

//STATE
class State1;
class State2;

//STATE_MACHINE
struct BaseStateMachine
    : public smacc::SmaccStateMachineBase<BaseStateMachine, State1>{

  BaseStateMachine(my_context ctx, smacc::SignalDetector *signalDetector)
      : smacc::SmaccStateMachineBase<BaseStateMachine, State1>(ctx, signalDetector){
  }
  
  virtual void onInitialize() override{
  }

};

struct State1
    : smacc::SmaccState<State1, BaseStateMachine>{

  smacc::SmaccMoveBaseActionClient *moveBaseClient_;

  typedef mpl::list<sc::transition<smacc::EvActionResult<smacc::SmaccMoveBaseActionClient>, State2>> reactions;  

  using SmaccState::SmaccState;
  void onInitialize(){
    ROS_INFO("Entering State1");
    this->requiresComponent(moveBaseClient_);
    goToEndPoint();
  }

  void goToEndPoint(){
    geometry_msgs::PoseStamped radialStartPose;
    
    smacc::SmaccMoveBaseActionClient::Goal goal;
    goal.target_pose.header.stamp = ros::Time::now();

    goal.target_pose = radialStartPose;
    goal.target_pose.pose.position.x = 10;
    goal.target_pose.pose.position.y = 10;
    goal.target_pose.pose.orientation =
    tf::createQuaternionMsgFromRollPitchYaw(0, 0, M_PI);

    moveBaseClient_->sendGoal(goal);
  }
};

struct State2
    : smacc::SmaccState<State2, BaseStateMachine>{

  using SmaccState::SmaccState;
  void onInitialize(){
    ROS_INFO("Entering State2");
  }
};
}//namespace smacc_tutorial

int main(int argc, char **argv){
  ros::init(argc, argv, "tutorial");
  ros::NodeHandle nh;

  smacc::run<smacc_tutorial::BaseStateMachine>();
}
``` 
Some code are changed from tutorial.

pabloinigoblasco, 2019-11-25:
Hello.
We are in a process of updating the API so there are a few obsolete functionalities in your code. 

A few notes about the issues in your code:

 - We replaced the concept of component for the concept of client (requiresClient instead of requiresComponent). The concept of component still exists but it is something more internals for subparts of a client.

 - Clients must be created in orthogonals, your code did not have any orthogonal. I added that functionality in the code showed below.

- Another issue is that you used sc::transitions (boost::statechart transitions). That it is not used anymore. Instead, we have the smacc::transition class.

- Finally, this is not mandatory but it is our convention: we usually put the "hard code" inside substateBehaviors instead of directly in States. You can use many substateBehaviors inside a state. Each substateBehavior goes usually inside an orthogonal (ie: Navigation Orthogonal)

In that substateBehavior you can also use the requiresClient functionality. 
We keep using the state class essentially for the transition table declaration and the configuration of orthogonals and substates.
To do that we also use instead of methods onEntry or onInitialize we use the "static void onDefinition()" method to configure behaviors because this improve the visualization with the smacc_viewer. The main difference is that onDefinition of all states executes at the begining of the process while onInitialize and onEntry  execute when you are entering in that state.

Below I show you  the code adapted to the latest smacc code:

```
#include <smacc/smacc.h>
#include <smacc_navigation_plugin/move_base_to_goal.h>
#include <tf/transform_datatypes.h>

namespace smacc_tutorial
{

//STATE
class State1;
class State2;
//--------------------------------------------------------------------
class NavigationOrthogonal : public smacc::Orthogonal
{
public:
    virtual void onInitialize() override
    {
        auto *client = this->createClient<smacc::SmaccMoveBaseActionClient>();
        client->name_ = "move_base";
        client->initialize();
    }
};

//--------------------------------------------------------------------
//STATE_MACHINE
struct BaseStateMachine
    : public smacc::SmaccStateMachineBase<BaseStateMachine, State1>
{
    using SmaccStateMachineBase::SmaccStateMachineBase;

    virtual void onInitialize() override
    {
        this->createOrthogonal<NavigationOrthogonal>();
    }
};

//--------------------------------------------------------------------
struct SbState1 : public smacc::SmaccSubStateBehavior
{
    smacc::SmaccMoveBaseActionClient *moveBaseClient_;

    virtual void onEntry() override
    {
        ROS_INFO("Entering State1");
        this->requiresClient(moveBaseClient_);
        goToEndPoint();
    }

    void goToEndPoint()
    {
        geometry_msgs::PoseStamped radialStartPose;

        smacc::SmaccMoveBaseActionClient::Goal goal;
        goal.target_pose.header.stamp = ros::Time::now();

        goal.target_pose = radialStartPose;
        goal.target_pose.pose.position.x = 10;
        goal.target_pose.pose.position.y = 10;
        goal.target_pose.pose.orientation =
            tf::createQuaternionMsgFromRollPitchYaw(0, 0, M_PI);

        moveBaseClient_->sendGoal(goal);
    }
};
//--------------------------------------------------------------------

struct State1
    : smacc::SmaccState<State1, BaseStateMachine>
{
    typedef mpl::list<smacc::transition<smacc::EvActionSucceded<smacc::SmaccMoveBaseActionClient>, State2>> reactions;

    using SmaccState::SmaccState;

    static void onDefinition()
    {
        static_configure<NavigationOrthogonal, SbState1>();
    }

    void onInitialize()
    {
    }
};
//--------------------------------------------------------------------
struct State2
    : smacc::SmaccState<State2, BaseStateMachine>
{
    using SmaccState::SmaccState;
    void onInitialize()
    {
        ROS_INFO("Entering State2");
    }
};
} //namespace smacc_tutorial

//--------------------------------------------------------------------
int main(int argc, char **argv)
{
    ros::init(argc, argv, "tutorial");
    ros::NodeHandle nh;

    smacc::run<smacc_tutorial::BaseStateMachine>();
}
```
I already tested this code compiles and I uploaded it as a tutorial example to the repo. It may also be useful for you to have a look to the cmakelist.txt file and the package.xml.

https://github.com/reelrbtx/SMACC/tree/master/smacc_sm_examples/sm_smacc_tutorial_examples

closed 2020-01-13 by reelrobotics