robosoft-ai/SMACC #17: SMACC vs BT

miccol, 2020-07-15:
Hi, I would like to start a discussion with SMACC vs BTs and have the opportunity to show my point. I am afraid you have been using BTs in the wrong context.

I consider the points that you raised up in the webpage [here](http://smacc.ninja/smacc-vs-behavior-trees/):

#### Orthogonality/Concurrency

I agree that having concurrent actions that need to be synchronized represents a challenge (and it is my current research interest now), however, I fail to see where SMACC is better than BTs in these contexts. In particular, you clam:

> What happens when you have 26 orthogonals, corresponding to 26 physical subsystems, say on a military uav. Are you going to run 26 processes and deal with all the interprocess communication? How would you debug such a system, and the inevitable race conditions that would almost certainly arise?
>
> It is my firm belief that this will not work because it will not scale

Will it scale with SMACC?

#### Recovery Modes/States

- How do you handle recovery sequences?

  - I need an example here, but I would put a Fallback operator of the two subtrees: the one for the nominal behavior and the one for the recovery behavior. But with an example, I think I can be more precise.
  - How would you with SMACC?

- Do you put them in the beginning of the cycle, or the end?

  - I would like to see an example (also with the SMACC design).

- if it’s at the beginning, how is this going to be fast, since you loop through it every time?

  - I am not sure what do you mean by that. We can consider a recovery behavior as a subtree (with a higher priority than the nominal behavior) with guarding conditions as the BT. Where the guarding conditions represent the conditions under which the recovery routine has to kick in.

- How do you handle orthogonals again? – see above

  - The problem is when you have non-orthogonal concurrent executions.

    

- Where are the robots?

  - I also had issues with finding public references for that. Companies don't expose their knowhow. I found a few examples:

    - Boston Dynamics

      - https://dev.bostondynamics.com/docs/concepts/autonomy/missions_service

    - Rethink Robotics

      - http://learn.skillman.eu/pluginfile.php/1195/mod_resource/content/0/Intera_5_User_Guide_Getting_Started_20170302_RevB.pdf

    - ROS2 Navigation stack

      - https://navigation.ros.org/

    - R1 in different fetch-and-pour task.

      - https://carve-robmosys.github.io/scenario1/
      - https://carve-robmosys.github.io/scenario2/
      - https://carve-robmosys.github.io/scenario3/

    - Robotic Behavior in RobMoSys using Behavior Trees and the SmartMDSD Toolchain (MOOD2be ITP)

      - https://www.youtube.com/watch?v=_EwNZG5Xo1k

    - COSTAR

      - https://www.youtube.com/watch?v=y4XkN48-b9k
      - https://www.youtube.com/watch?v=PXTdNY526S4

    - Learning BTs from demonstration

      - https://www.youtube.com/watch?v=OFL5x5vuPNc

#### Readability

From the web page

> Try an experiment out for yourself. Find someone, who has never taken a computer science course in their life, and ask them a question like…
>
> “How do you bake a cake?”

I think you are using the wrong example here. I would not use a BT for a static sequence of actions (or I would use simple BT formed by a sequence with memory with all the actions or a FSM). 

#### Performance

> And, I don’t think the condition based flow (polling) lends itself to high performance, vis a vis an event-based system. Particularly in asynchronous environments.

That depends on the implementation. However, also in FSM, you need to poll conditions to verify if it has to jump to another state (with the additional problem that you must ensure that the conditions are mutually exclusive, otherwise you get an undefined behavior).

#### States really do exist

Again from your page:

> In the final analysis, it should be obvious that although they are a mental construct, states really do exist…
>
> - In an airplane, you’re either in the air or on the ground
> - A car is either moving, or its not
> - A handgun is either loaded, or it’s not
> - Your mother-in-law has either joined you for dinner, or she hasn’t

You actually stated conditions that could be either true or false.

In BT you don't pretend that state does not exist, you consider only what matters. Take your first example "In an airplane, you’re either in the air or on the ground": you check the condition "if it is in the air" and not really the state. Depending on the state space representation, if you are sitting in the Business or Economy class can be represented as different states. 

#### Conclusions

To conclude, I think SMACC and BT have to be used for different problems (BT to model the robot deliberation that needs to react to several system changes and SMACC to model linear action sequencing). However, since you raised up some downsides of BTs, I would be really happy to see how SMACC outperforms them.

Regards

Michele Colledanchise

SteveMacenski, 2021-05-04:
@brettpac @reelrobotics @pabloinigoblasco I think enough time has passed since our last discourse discussion to revive this with more friendly intentions. 

https://smacc.dev/smacc-vs-behavior-trees/ this page is very aggressive and I don't think is in the spirit of open-source nor in the best interest of either BT or SMACC users. Striking out commentary doesn't remove it and this is propagating misunderstandings that @miccol has very calmly tried to clarify. 

I have no issue with SMACC project trying to offer a comparison w.r.t. BTs that you see as not the right solution, but I would propose that you take another look at them, with the resource of Michele and I, to make a more fair and productive comparison that isn't putting down others (but rather lifting up yourself)

reelrobotics, 2021-05-05:
Hi Steve, 
I'm open to this. Been on my todo list.

I'll reach out offline.

closed 2022-12-30 by brettpac