robosoft-ai/SMACC #18: CMake Error

davideferrari95, 2020-08-17:
Following the [Getting Started Tutorial](https://smacc.ninja/getting-started/), when I have to compile the workspace (after using `rosdep install`) I get the following error:

`CMake Error at SMACC/smacc_sm_reference_library/sm_ridgeback_barrel_search_2/CMakeLists.txt:92 (add_executable):
  add_executable cannot create target "opencv_perception_node" because
  another target with the same name already exists.  The existing target is
  an executable created in source directory
  "/home/davide/catkin_ws/src/SMACC/smacc_sm_reference_library/sm_ridgeback_barrel_search_1".
  See documentation for policy CMP0002 for more details.
`

And tons of related error cause `opencv_perception_node` is not build.

I'm running ROS Melodic, using an empty workspace.

reelrobotics, 2020-08-18:
Hi Davide,
Thank you for submitting this issue.

I wasn't able to directly duplicate the same error, but we did come across kind of a weird compilation errors related to a recent ros-melodic update.

Here is a short video, that shows how we fixed it (by updating various packages using synaptic). Please try this and let me know if it works for you...

[(https://vimeo.com/448792485)]

Thank you,

davideferrari95, 2020-08-18:
I tried to follow the video tutorial but, when I search in Synaptic `ros-melodic`, I don't have any packages marked with exclamation point (I guess they are all updated already).

So `catkin_make` continue to fail with the same type of error:

`CMake Error at SMACC/smacc_sm_reference_library/sm_ridgeback_barrel_search_2/CMakeLists.txt:92 (add_executable):
add_executable cannot create target "opencv_perception_node" because
another target with the same name already exists. The existing target is
an executable created in source directory
"/home/davide/catkin_ws/src/SMACC/smacc_sm_reference_library/sm_ridgeback_barrel_search_1".
See documentation for policy CMP0002 for more details.`

Analyzing the error I noticed that the problem is given by the existence of targets with the same name in different packages:

For example `opencv_perception_node` is in both the `sm_ridgeback_barrel_search_1` package and the `sm_ridgeback_barrel_search_2` package.

The same thing happens for `temperature_sensor_node` and `led_action_server_node` in `sm_ridgeback_floor_coverage_static_1` package and for `temperature_sensor_node_smdbsb` and `led_action_server_node_smdbsb` in `sm_ridgeback_floor_coverage_dynamic_1` package.

I found a solution that allows the build of the workspace, but I don't know what effects it can have and if it can cause problems on the correct functioning of the library:

Modifying the CMakeLists of these three packages, changing the name of the executables that create the error I was able to build the workspace. For example, in file `sm_ridgeback_floor_coverage_dynamic_1/CMakeLists.txt` I renamed nodes `temperature_sensor_node_smdbsb` and `led_action_server_node_smdbsb` by appending "_1".

```
 add_executable(temperature_sensor_node_smdbsb_1 servers/temperature_sensor_node/src/temperature_sensor_node.cpp)
 set_target_properties(temperature_sensor_node_smdbsb_1 PROPERTIES OUTPUT_NAME "temperature_sensor_node")

 target_link_libraries(temperature_sensor_node_smdbsb_1
   ${catkin_LIBRARIES}
 )

add_executable(led_action_server_node_smdbsb_1 servers/led_action_server/src/led_action_server_node.cpp)
set_target_properties(led_action_server_node_smdbsb_1 PROPERTIES OUTPUT_NAME "led_action_server_node")

 target_link_libraries(
   led_action_server_node_smdbsb_1
   ${catkin_LIBRARIES}
)

add_dependencies(led_action_server_node_smdbsb_1 ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS} ${PROJECT_NAME}_generate_messages_cpp)

#############
## Install ##
#############

# all install targets should use catkin DESTINATION variables
# See http://ros.org/doc/api/catkin/html/adv_user_guide/variables.html

## Mark executable scripts (Python etc.) for installation
## in contrast to setup.py, you can choose the destination
install(PROGRAMS
   servers/service_node_3/src/service_node_3.py
   DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

## Mark executables for installation
## See http://docs.ros.org/melodic/api/catkin/html/howto/format1/building_executables.html
install(TARGETS 
  ${PROJECT_NAME}
   temperature_sensor_node_smdbsb_1
   led_action_server_node_smdbsb_1
   RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

reelrobotics, 2020-08-18:
Ah, I see. You used catkin_make.
You bring up a good point regarding your fix, we'll look at getting this working with catkin_make but it will take a little time (Sept. 1-5).

Any particular reason you're not using catkin build?

davideferrari95, 2020-08-18:
> You bring up a good point regarding your fix, we'll look at getting this working with catkin_make but it will take a little time (Sept. 1-5).

Perfect, I'll wait for the fix. Today I tried to launch sm_atomic after my build workaround and it works good.

> Any particular reason you're not using catkin build?

I learned ROS following the [Official ROS Tutorials](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment) so I have always used catkin_make.

Another question, is there any beginner tutorial for this package? I need to understand how to implement the sm in my code. Thanks

reelrobotics, 2020-08-18:
Ok, we'll plan on fixing this. 
But I highly recommend catkin build. It's better.

Regarding a beginner tutorial, I don't know if we have what you're looking for. But I'll contact you off this topic to find out more about what you're trying to do. That way this issue will remain compilation focused.

davideferrari95, 2020-08-18:
> But I highly recommend catkin build. It's better.

Good I'll try it for maybe a future work, now all the codes are built with catkin_make.

> Regarding a beginner tutorial, I don't know if we have what you're looking for. But I'll contact you off this topic to find out more about what you're trying to do. That way this issue will remain compilation focused.

Okay, thanks.

davideferrari95, 2020-09-06:
Is there any news for solving the problem with catkin_make? I am waiting to be able to implement the library inside my code.

You can contact me privately at the email address on my github profile, I would also need some tutorials of use.

I await news, Davide

pabloinigoblasco, 2020-09-08:
@davideferrari95 

Thanks for your suggestion, SMACC should build properly with catkin_make now (commit https://github.com/reelrbtx/SMACC/commit/f8e3be537c92009cc7759b610828f6f9e8557566)

Please, check it out.

davideferrari95, 2020-09-09:
I did the git clone of the latest release (commit [3e25e17](https://github.com/reelrbtx/SMACC/commit/3e25e17fb20d7dacb4ffcad1d6652630bdac8a25)) and the catkin_make worked fine.

Thanks for the support, I close the issue.

closed 2020-09-09 by davideferrari95