robosoft-ai/SMACC #19: Compiling SMACC

evangelosc, 2020-10-06:
Hi guys,

I tried to follow your Getting Started video, but I am getting a lot of problems at the compiling part.
Please see my log below and let me know if you've encountered similar problems.

**Catkin Build Summary**
[build] Summary: 4 of 50 packages succeeded.                                   
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 42 packages were abandoned.                               
[build]   Failed:    4 packages failed.                                        
[build] Runtime: 1 minute and 34.7 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

**Actual Log**
[build] Found '49' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                                [ 2.0 seconds ]
Starting  >>> forward_global_planner                                           
Starting  >>> move_eye_client                                                  
Starting  >>> pure_spinning_local_planner                                      
Starting  >>> smacc_msgs                                                       
Finished  <<< move_eye_client                                      [ 2.0 seconds ]
Finished  <<< smacc_msgs                                           [ 8.9 seconds ]
Starting  >>> smacc                                                            
Starting  >>> smacc_runtime_test                                               
Starting  >>> smacc_rviz_plugin                                                
_______________________________________________________________________________
Errors     << pure_spinning_local_planner:make /home/evangelos/Desktop/catkin_smacc/logs/pure_spinning_local_planner/build.make.000.log
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/pure_spinning_local_planner.dir/src/pure_spinning_local_planner/pure_spinning_local_planner.cpp.o] Error 4
make[1]: *** [CMakeFiles/pure_spinning_local_planner.dir/all] Error 2
make: *** [all] Error 2
cd /home/evangelos/Desktop/catkin_smacc/build/pure_spinning_local_planner; catkin build --get-env pure_spinning_local_planner | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << pure_spinning_local_planner:make                     [ Exited with code 2 ]
Failed    <<< pure_spinning_local_planner                          [ 19.4 seconds ]
Abandoned <<< backward_global_planner                              [ Unrelated job failed ]
Abandoned <<< backward_local_planner                               [ Unrelated job failed ]
Abandoned <<< forward_local_planner                                [ Unrelated job failed ]
Abandoned <<< battery_monitor_client                               [ Unrelated job failed ]
Abandoned <<< eg_conditional_generator                             [ Unrelated job failed ]
Abandoned <<< eg_random_generator                                  [ Unrelated job failed ]
Abandoned <<< keyboard_client                                      [ Unrelated job failed ]
Abandoned <<< microstrain_mips_client                              [ Unrelated job failed ]
Abandoned <<< move_base_z_client_plugin                            [ Unrelated job failed ]
Abandoned <<< move_group_interface_client                          [ Unrelated job failed ]
Abandoned <<< multirole_sensor_client                              [ Unrelated job failed ]
Abandoned <<< ros_publisher_client                                 [ Unrelated job failed ]
Abandoned <<< ros_timer_client                                     [ Unrelated job failed ]
Abandoned <<< sm_atomic_cb                                         [ Unrelated job failed ]
Abandoned <<< sm_coretest_transition_speed_1                       [ Unrelated job failed ]
Abandoned <<< sm_coretest_x_y_1                                    [ Unrelated job failed ]
Abandoned <<< sm_coretest_x_y_2                                    [ Unrelated job failed ]
Abandoned <<< sm_coretest_x_y_3                                    [ Unrelated job failed ]
Abandoned <<< sm_fetch_screw_loop_1                                [ Unrelated job failed ]
Abandoned <<< sm_fetch_six_table_pick_n_sort_1                     [ Unrelated job failed ]
Abandoned <<< sm_fetch_two_table_pick_n_place_1                    [ Unrelated job failed ]
Abandoned <<< sm_fetch_two_table_whiskey_pour                      [ Unrelated job failed ]
Abandoned <<< sm_ridgeback_barrel_search_1                         [ Unrelated job failed ]
Abandoned <<< sm_ridgeback_barrel_search_2                         [ Unrelated job failed ]
Abandoned <<< sm_update_loop                                       [ Unrelated job failed ]
Abandoned <<< sm_atomic                                            [ Unrelated job failed ]
Abandoned <<< sr_all_events_go                                     [ Unrelated job failed ]
Abandoned <<< sm_calendar_week                                     [ Unrelated job failed ]
Abandoned <<< sm_ferrari                                           [ Unrelated job failed ]
[... 58 lines cut ...]
Failed     << smacc:make                                           [ Exited with code 2 ]
Failed    <<< smacc                                                [ 1 minute and 18.4 seconds ]
Finished  <<< smacc_runtime_test                                   [ 1 minute and 21.6 seconds ]
_______________________________________________________________________________
Errors     << smacc_rviz_plugin:make /home/evangelos/Desktop/catkin_smacc/logs/smacc_rviz_plugin/build.make.000.log
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/smacc_rviz_plugin.dir/src/smacc_rviz_display.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/smacc_rviz_plugin.dir/smacc_rviz_plugin_autogen/mocs_compilation.cpp.o] Error 4
make[1]: *** [CMakeFiles/smacc_rviz_plugin.dir/all] Error 2
make: *** [all] Error 2
cd /home/evangelos/Desktop/catkin_smacc/build/smacc_rviz_plugin; catkin build --get-env smacc_rviz_plugin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << smacc_rviz_plugin:make                               [ Exited with code 2 ]
Failed    <<< smacc_rviz_plugin                                    [ 1 minute and 23.4 seconds ]

Best regards,
Evangelos.

evangelosc, 2020-10-06:
I run on a VirtualBox Ubuntu 18.04.5 LTS with melodic.

evangelosc, 2020-10-06:
I am not getting that error when I compile the SMACC on a 32gb RAM computer, however it crashes on my laptop which has a 8gb RAM. What's the min RAM req?

pabloinigoblasco, 2020-10-06:
That is a typical problem of the g++ compiler when it is reaching the limits of memory.
You need more memory. 

As a workaround you can at least decrease the compilation parallelism ( catkin_make -j 1) and less memory will be used.

evangelosc, 2020-10-06:
Thanks for the response. I imagined that it has to do with that.

closed 2020-10-06 by evangelosc