robosoft-ai/SMACC #20: States, Topics & Data Structures

evangelosc, 2020-10-14:
Hi,

I had a look at sm_fetch_six_table_pick_n_sort_1 and the sm_atomic of the reference library and want to ask you a few questions about their functionalities.
1) _sm_atomic_: I want to create a naive publisher (e.g. publish an int array) in state_1, where should I do it? As the state comes in the form of the header file, I cannot initialize a ros node handler there, right? 
2) _sm_fetch_six_table_pick_n_sort_1_ : You store the moveit plan in a moveit_msgs/DisplayTrajectory.msg, however moveit usually saves it in the RobotTrajectory component of the above data structure. Why do you use the above message and not the RobotTrajectory one? I'm referring particularly at the /move_group/display_generated_plan topic.
3) _sm_fetch_six_table_pick_n_sort_1_ : How does the moveit client translate the generated plan into joint movement in you gazebo simulation? Can you please point me at the specific part of the code?

Thank you!

evangelosc, 2020-10-15:
Regarding point 1, I created a publisher orthogonal in state_2 of the sm_atomic and got a few weird ros errors. Please see log below. The errors kicks at the line where I configure the topic of the publisher. Should the configuration be on different way?

`void runtimeConfigure()
    {
        ClRosPublisher pub;
        pub.configure("/newTopic");
    }`

Thanks!

/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc_sm_reference_library/sm_atomic/include/sm_atomic/states/st_state_2.h:27:34:   required from here
/opt/ros/melodic/include/ros/message_traits.h:120:28: error: ‘__s_getMD5Sum’ is not a member of ‘const char*’
     return M::__s_getMD5Sum().c_str();
            ~~~~~~~~~~~~~~~~^~
/opt/ros/melodic/include/ros/message_traits.h: In instantiation of ‘static const char* ros::message_traits::DataType<M>::value() [with M = const char*]’:
/opt/ros/melodic/include/ros/message_traits.h:236:104:   required from ‘const char* ros::message_traits::datatype() [with M = const char*]’
/opt/ros/melodic/include/ros/advertise_options.h:93:43:   required from ‘void ros::AdvertiseOptions::init(const string&, uint32_t, const SubscriberStatusCallback&, const SubscriberStatusCallback&) [with M = const char*; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int; ros::SubscriberStatusCallback = boost::function<void(const ros::SingleSubscriberPublisher&)>]’
/opt/ros/melodic/include/ros/node_handle.h:252:7:   required from ‘ros::Publisher ros::NodeHandle::advertise(const string&, uint32_t, bool) [with M = const char*; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int]’
/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc/include/smacc/client_bases/smacc_publisher_client.h:47:12:   required from ‘void smacc::client_bases::SmaccPublisherClient::configure(std::__cxx11::string) [with MessageType = const char*; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc_client_library/ros_publisher_client/include/ros_publisher_client/cl_ros_publisher.h:19:53:   required from ‘void cl_ros_publisher::ClRosPublisher::configure(MessageType) [with MessageType = const char*]’
/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc_sm_reference_library/sm_atomic/include/sm_atomic/states/st_state_2.h:27:34:   required from here
/opt/ros/melodic/include/ros/message_traits.h:137:30: error: ‘__s_getDataType’ is not a member of ‘const char*’
     return M::__s_getDataType().c_str();
            ~~~~~~~~~~~~~~~~~~^~
/opt/ros/melodic/include/ros/message_traits.h: In instantiation of ‘static const char* ros::message_traits::Definition<M>::value() [with M = const char*]’:
/opt/ros/melodic/include/ros/message_traits.h:245:106:   required from ‘const char* ros::message_traits::definition() [with M = const char*]’
/opt/ros/melodic/include/ros/advertise_options.h:94:55:   required from ‘void ros::AdvertiseOptions::init(const string&, uint32_t, const SubscriberStatusCallback&, const SubscriberStatusCallback&) [with M = const char*; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int; ros::SubscriberStatusCallback = boost::function<void(const ros::SingleSubscriberPublisher&)>]’
/opt/ros/melodic/include/ros/node_handle.h:252:7:   required from ‘ros::Publisher ros::NodeHandle::advertise(const string&, uint32_t, bool) [with M = const char*; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int]’
/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc/include/smacc/client_bases/smacc_publisher_client.h:47:12:   required from ‘void smacc::client_bases::SmaccPublisherClient::configure(std::__cxx11::string) [with MessageType = const char*; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc_client_library/ros_publisher_client/include/ros_publisher_client/cl_ros_publisher.h:19:53:   required from ‘void cl_ros_publisher::ClRosPublisher::configure(MessageType) [with MessageType = const char*]’
/home/evangelos/Desktop/catkin_ws/src/SMACC/smacc_sm_reference_library/sm_atomic/include/sm_atomic/states/st_state_2.h:27:34:   required from here
/opt/ros/melodic/include/ros/message_traits.h:154:39: error: ‘__s_getMessageDefinition’ is not a member of ‘const char*’
     return M::__s_getMessageDefinition().c_str();
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
SMACC/smacc_sm_reference_library/sm_atomic/CMakeFiles/sm_atomic_node.dir/build.make:62: recipe for target 'SMACC/smacc_sm_reference_library/sm_atomic/CMakeFiles/sm_atomic_node.dir/src/sm_atomic_node.cpp.o' failed
make[2]: *** [SMACC/smacc_sm_reference_library/sm_atomic/CMakeFiles/sm_atomic_node.dir/src/sm_atomic_node.cpp.o] Error 1
CMakeFiles/Makefile2:5487: recipe for target 'SMACC/smacc_sm_reference_library/sm_atomic/CMakeFiles/sm_atomic_node.dir/all' failed
make[1]: *** [SMACC/smacc_sm_reference_library/sm_atomic/CMakeFiles/sm_atomic_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

pabloinigoblasco, 2020-10-15:
> Hi,
>
> I had a look at sm_fetch_six_table_pick_n_sort_1 and the sm_atomic of the reference library and want to ask you a few questions about their functionalities.
>
> 1. _sm_atomic_: I want to create a [...]

Hello evangelosc. Some fast answers.
1. Clients are located in orthogonals, so they are created at the very beginning. We consider clients objects that live for the whole execution.
Nonetheless, standard ros publishers can be created in states or inside clientbehaviors (even if they are in include files). However, in these cases the lifetime of the publishers will be bounded.

2. Could you please reference exactly which the file and line where you say that happens?

3. The robot arm is controlled by a remote moveit_node. In smacc, we consider that remote node a client ([ClMoveGroup](https://github.com/reelrbtx/SMACC/blob/master/smacc_client_library/move_group_interface_client/src/move_group_interface_client/cl_movegroup.cpp)). In our local SMACC node (or program) we use the concept of "client behaviors" to comunicate with the remote moveit_node and do certain tasks. These "client behaviors" are essentially "small programs" that run for some "scoped time" in a state. When we leave the state, that client behavior stop working. 

We have a bunch of client behaviors implemented for moveit, have a look to: 
- https://github.com/reelrbtx/SMACC/tree/master/smacc_client_library/move_group_interface_client/include/move_group_interface_client/client_behaviors
- https://github.com/reelrbtx/SMACC/tree/master/smacc_client_library/move_group_interface_client/src/move_group_interface_client/client_behaviors

pabloinigoblasco, 2020-10-15:
> Regarding point 1, I created a publisher orthogonal in state_2 of the sm_atomic and got a few weird ros errors. Please see log below. The errors kicks at the line where I configure the topic of the [...]

In Smacc, clients are considered "long lifetime objects", so they do not belong to any state. 
Instead the belong to "an orthogonal" (another long lifetime object).

Have a look how we create a basic publisher [here](https://github.com/reelrbtx/SMACC/blob/master/smacc_sm_reference_library/sm_dance_bot/include/sm_dance_bot/orthogonals/or_updatable_publisher.h). This example shows how we define the "OrUpdatablePublisher" orthogonal. We create inside a publisher client (ClRosPublisher). This happens during the initialization of the state machine.

```
#include <ros_publisher_client/cl_ros_publisher.h>
#include <smacc/smacc_orthogonal.h>

namespace sm_dance_bot
{
class OrUpdatablePublisher : public smacc::Orthogonal<OrUpdatablePublisher>
{
public:
    virtual void onInitialize() override
    {
        auto publisherClient_ = this->createClient<cl_ros_publisher::ClRosPublisher>();
        publisherClient_->configure<std_msgs::String>("/updatable_string_publisher_out");
    }
};
} // namespace sm_dance_bot
```

From this initial moment any state, behavior, component or even another client can find this publisher client to interact.
For example, you could do from a client behavior located in the same orthogonal the following.

```
 ClRosPublisher* pub;
 this->requiresClient(pub);
 pub->publish(msg);
```

evangelosc, 2020-10-17:
Hi @pabloinigoblasco, Thank you very much for your clarification. Appreciate all your help and the resources you provided.

closed 2020-10-22 by pabloinigoblasco