robosoft-ai/SMACC #21: Not compiling the sm_atomic and other code examples

linuxsocke, 2020-10-22:
Hello, 

Somehow when running `catkin_make` I have compiling errors with the sm_atomic example and probably the others:

In file included from /usr/include/boost/statechart/fifo_scheduler.hpp:13,
                 from /usr/include/boost/statechart/asynchronous_state_machine.hpp:12,
                 from /home/user/Workspace/ws_catkin/src/SMACC/smacc/include/smacc/common.h:11,
                 from /home/user/Workspace/ws_catkin/src/SMACC/smacc/include/smacc/smacc.h:7,
                 from /home/user/Workspace/ws_catkin/src/SMACC/smacc_sm_reference_library/sm_atomic/include/sm_atomic/sm_atomic.h:1,
                 from /home/user/Workspace/ws_catkin/src/SMACC/smacc_sm_reference_library/sm_atomic/src/sm_atomic_node.cpp:1:

/usr/include/boost/statechart/processor_container.hpp: In instantiation of ‘void boost::statechart::processor_container<Scheduler, WorkItem, Allocator>::create_processor_impl1(const processor_holder_ptr_type&, const boost::statechart::processor_container<Scheduler, WorkItem, Allocator>::processor_context&, Arg1) [with Processor = sm_atomic::SmAtomic; Arg1 = smacc::SignalDetector*; Scheduler = boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<void> >, std::allocator<void> >; WorkItem = boost::function0<void>; Allocator = std::allocator<void>; boost::statechart::processor_container<Scheduler, WorkItem, Allocator>::processor_holder_ptr_type = boost::shared_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<void> >, std::allocator<void> > > > >]’:
/usr/include/boost/statechart/processor_container.hpp:136:20:   required from ‘WorkItem boost::statechart::processor_container<Scheduler, WorkItem, Allocator>::create_processor(boost::statechart::processor_container<Scheduler, WorkItem, Allocator>::processor_handle&, Scheduler&, Arg1) [with Processor = sm_atomic::SmAtomic; Arg1 = smacc::SignalDetector*; Scheduler = boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<void> >, std::allocator<void> >; WorkItem = boost::function0<void>; Allocator = std::allocator<void>; boost::statechart::processor_container<Scheduler, WorkItem, Allocator>::processor_handle = boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<void> >, std::allocator<void> > > > >]’
/usr/include/boost/statechart/fifo_scheduler.hpp:62:17:   required from ‘boost::statechart::fifo_scheduler<FifoWorker, Allocator>::processor_handle boost::statechart::fifo_scheduler<FifoWorker, Allocator>::create_processor(Arg1) [with Processor = sm_atomic::SmAtomic; Arg1 = smacc::SignalDetector*; FifoWorker = boost::statechart::fifo_worker<std::allocator<void> >; Allocator = std::allocator<void>; boost::statechart::fifo_scheduler<FifoWorker, Allocator>::processor_handle = boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<void> >, std::allocator<void> > > > >]’
/home/linuxsocke/Nextcloud/Workspace/ws_catkin/src/SMACC/smacc/include/smacc/smacc_signal_detector.h:89:42:   required from ‘void smacc::run() [with StateMachineType = sm_atomic::SmAtomic]’
/home/linuxsocke/Nextcloud/Workspace/ws_catkin/src/SMACC/smacc_sm_reference_library/sm_atomic/src/sm_atomic_node.cpp:9:37:   required from here
/usr/include/boost/statechart/processor_container.hpp:324:44: 
**error:** no matching function for call to ‘sm_atomic::SmAtomic::SmAtomic(const boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<void> >, std::allocator<void> >, boost::function0<void>, std::allocator<void> >::processor_context&, smacc::SignalDetector*&)’
  324 |       *pProcessor = processor_holder_type( **new Processor( context, arg1 )** );

Is this version issue with boost?

I am using Ubnunt 20.04 with ros-noetic.

pabloinigoblasco, 2020-10-22:
Hello @linuxsocke, thanks for trying smacc and reporting this.

I exactly know what it is happening. 

As you say, the issue is in libboost-all-dev. There are some syntactical changes in the new version of boost-statechart that is breaking the ubuntu 18.04 build in 20.04. The master branch is right now "tracking" ros-melodic/ubuntu18.04.

I know that issue, because I already fixed these issues in the effort we are doing portting the library to ros2. Please check: https://github.com/droidware-ai/SMACC2

That version of smacc works on ubuntu 20.04 - ros-foxy and I had to fix there all these issues you are mentioning. Alll of them are mostly related with the replacement of one allocator std::allocator -> to std::allocator<boost::statechart::none>.  
Right the work for the migration to noetic paused at least until we complete the migration to ros2 (then we will decide what to do with this).

If you want to try to solve this, please check the differences related with SmaccAllocator and std::allocator<boost::statechart::none> in the smacc2 repository:

https://github.com/droidware-ai/SMACC2/blob/master/smacc2/include/smacc/smacc_fifo_worker.h
https://github.com/droidware-ai/SMACC2/blob/master/smacc2/include/smacc/common.h
https://github.com/droidware-ai/SMACC2/blob/master/smacc2/include/smacc/introspection/introspection.h

Please, feel free to make a pull request if you want to fix this issue. It shouldn't be hard to move from melodic to noetic (altough we have many projects and booststate errors are very verbosy and scary).

It would be great if someone could check at least that the basic core packages (smacc) build in noetic. if clients or examples fails we can always deactivate them by the moment.

linuxsocke, 2020-10-23:
Ok, thank you for the fast response! So I will close this issue.

closed 2020-10-23 by linuxsocke