robosoft-ai/SMACC #22: SMACC for noetic

Soft-illusion, 2021-02-08:
ROSDEP does not install dependencies.

rosdep install --from-paths src --ignore-src -r -y

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
sm_fetch_six_table_pick_n_sort_1: Cannot locate rosdep definition for [fetch_navigation]
sm_dance_bot: Cannot locate rosdep definition for [ridgeback_navigation]
sm_dance_bot_2: Cannot locate rosdep definition for [ridgeback_navigation]
sm_ridgeback_barrel_search_2: Cannot locate rosdep definition for [ridgeback_navigation]
sm_fetch_two_table_pick_n_place_1: Cannot locate rosdep definition for [fetch_navigation]
sm_dance_bot_strikes_back: Cannot locate rosdep definition for [ridgeback_navigation]
sm_ridgeback_barrel_search_1: Cannot locate rosdep definition for [ridgeback_navigation]
microstrain_mips_client: Cannot locate rosdep definition for [microstrain_mips]
sm_fetch_two_table_whiskey_pour: Cannot locate rosdep definition for [fetch_navigation]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Any solution ?

pabloinigoblasco, 2021-02-08:
Hello.
At the current moment noetic is not supported, but it will be eventually. 

Nonetheless, there are some solutions.
- The fastest one is to remove all those packages that fail
- include the code of those submodules into your workspace and try to make them compile for noetic

All the packages that starts with sm_ are just examples, you also would need to remove the microstrain client.

After that I am aware that **another issue will show up**. There is some updates there are in the boost statechart library for ubuntu 20.04 (we already experimented that for our ros2 version we are developing).

It is an issue related with the replacement of a std::allocator<void>  in some clases of boost statechart with another class std::allocator<boost::statechart::none>.

That issue is more ore less explained in this other issue: https://github.com/reelrbtx/SMACC/issues/21#issuecomment-714615398

In resume.
- It is possible to do it
- It is in the TODO list
- Any pull request is welcome If you make it compile

Regards.

koyalbhartia, 2021-02-09:
Hi, it's me who opened the issue above via another account.

I made the following changes and SMACC compiles fine.
1. Changed the std::allocator to match as shown in these files:
https://github.com/droidware-ai/SMACC2/blob/master/smacc2/include/smacc/smacc_fifo_worker.h
https://github.com/droidware-ai/SMACC2/blob/master/smacc2/include/smacc/common.h
https://github.com/droidware-ai/SMACC2/blob/master/smacc2/include/smacc/introspection/introspection.h
2. Removed the packages move_group_interface_client, microstrain_mips_client
3. Removed all examples from smacc_sm_reference_library except sm_atomic.

**With a successful compilation and source, I launched the following "roslaunch sm_atomic sm_atomic.launch".** However, I get the following runtime test error. Does any package any further modification, or can I send a pull request and move forward with the implementation of SMACC for my application in ROS Noetic and Ubuntu 20.04..?

[image]

koyalbhartia, 2021-02-09:
I can send a pull request of the above changes if you wish to take a look and validate the same.

pabloinigoblasco, 2021-02-10:
Thank you @koyalbhartia . Yes please, make a pull request. I will review it.

koyalbhartia, 2021-02-10:
Created the pull request.

stoic-roboticist, 2021-02-11:
> Hi, it's me who opened the issue above via another account.
>
> I made the following changes and SMACC compiles fine.
>
>     1. Changed the std::allocator to match as shown in these files: [...]

I complied the code using your suggestions and it worked on my machine with ROS noetic. The error you see is due to exit(-1) being called in the smacc_runtime_test_node.cpp, which kills the node process and produces the error message you see. Maybe a cleaner way is needed when killing this test node.

pabloinigoblasco, 2021-02-11:
I created a new branch for noetic "noetic-devel"

It builds properly and saves many of the examples we had for melodic. (including navigation ridgeback ones via git submodules)
Master is still melodics but eventually noetic will become master.

Do not forget to pull submodules and also execute rosdep tools.

Thanks to everybody for the help and the pull requests you did.

Max80well80, 2021-02-17:
Hello, I am having some trouble getting noetic-devel to build. 

I may be making a stupid mistake so I tried to lay out exactly what I'm trying
inside my catkin_ws/src
1. Git clone git clone https://github.com/reelrbtx/SMACC.git
2. cd SMACC
3. git checkout noetic-devel
4. git submodule init
5. git submodule update
6. navigate back to catkin_ws and run rosdep install --from-paths src --ignore-src -r -y
7. catkin_make
its here the make process fails:

catkin_ws/src/SMACC/smacc_client_library/move_base_z_client/custom_planners/pure_spinning_local_planner/include/pure_spinning_local_planner/pure_spinning_local_planner.h:14:10: fatal error: pure_spinning_local_planner/PureSpinningLocalPlannerConfig.h: No such file or directory
   14 | #include <pure_spinning_local_planner/PureSpinningLocalPlannerConfig.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Did I miss a step?

pabloinigoblasco, 2021-02-17:
It is a bit weird. I am not sure what it is happening. It looks like the autogenerated header file for dynamic reconfigure was not properly detected.

I am not getting that issue on my computer.
Travis build (from a clean machine) also looks to be working
https://travis-ci.com/github/reelrbtx/SMACC/builds/217009917

are you totally updated in the branch? do you have installed dynamic reconfigure?

Max80well80, 2021-02-17:
Hi, I have the noetic-devel branch of dynamic reconfigure, should i be using the master for that?

I am also running this on 20.04.

pabloinigoblasco, 2021-02-17:
Everything should be running on ubuntu 20.04 and noetic-devel branch.
master is now based on melodic, so do not use that.

Nonetheless, I am researching about your issue, I did not forget it. Right now I am trying to make the debian files generation that should we working eventually. Maybe that could be enough for you.

If you really want to get the code working you may be interested in the docker image we use for the debian generation (that needs to download and compile the code) (still in progress)

Max80well80, 2021-02-17:
Thanks! I'll check out the docker image, I'm also going to scrub my setup and try again just in case i made a mistake somewhere.

Max80well80, 2021-02-17:
Alright well thankfully (but frustratingly for bug hunting) post scrub everything worked and built successfully. Not sure what I changed. For any future readers, make sure you have the noetic-devel of the dynamic reconfigure installed!

pabloinigoblasco, 2021-02-17:
Great.

I have been changing noetic these last few hours, maybe something was fixed.

In any case it looks like the debian generation for noetic is already working locally based on this Dockerfile configuration:
[https://github.com/reelrbtx/SMACC/blob/noetic-devel/smacc_ci/packagecloud_docker/Dockerfile](https://github.com/reelrbtx/SMACC/blob/noetic-devel/smacc_ci/packagecloud_docker/Dockerfile)

closed 2021-05-27 by pabloinigoblasco