robosoft-ai/SMACC #23: Pass information from one state to another

Alex-Beh, 2021-02-10:


reelrobotics, 2021-02-11:
Hi Alex,
Can you offer more specifics on what you're trying to do, or what type of data we're working with?
The reason I'm asking is that there are multiple ways of doing this.

One very simple mechanism is to simply use a "global" variable loaded into the state machine, which we've already got functions for. Try this github search to see what I'm talking about... https://github.com/reelrbtx/SMACC/search?q=getGlobalSMData&type=  
We don't have any examples in the reference library currently using this method, but it works.

Another option, say in the case that you only want data scoped to the lifetime of a group of states, but not the whole state machine is to use a super state or a mode state. We have lots of examples where we use them kind of like for loops,  and share data in that sense and they usually have a data field that at least stores how many times they've been through the loop.

But I need more information please;)...

pabloinigoblasco, 2021-02-12:
My two cent around. Very short.
I would say there are two main options:
- Using the StateMachine global dicctionary of variables (based on boost::any)
- Using objects with a long lifecyle (orthogonals, clients and components). Long lifecycle elements (orghogonals, clients, components) can be accessed freely from any state or state behavior.

From my understanding the best approach is using components. A component can be "custom structure" totally specific of your application and that can be hosted inside clients (that are like containers of components).

For example: "the navigation mission details" could be stored in the moveBaseClient.

Then, states or state behavior (short lifecycle) would read/write over that component to do/behave in some specific way based on that information.
--------
Another way of passing information between elements in smacs are events (that can contain information), but that is another animal to comunicate clients, behaviors and components with states (but not states with other states)

closed 2021-03-17 by reelrobotics