robosoft-ai/SMACC #24: SMACC Viewer For Ubuntu20.04

astroxyx, 2021-02-10:
hello ,now I can run smacc in 20.04 by delete some client and modify the boost.But I alse want to use SMACC Viewer in 20.04.Can you release version 20.04?

reelrobotics, 2021-02-15:
Hi Astroxyx,
We expect to have a version for 20.04 by the end of this week.

astroxyx, 2021-02-16:
Thank you very much.

astroxyx, 2021-02-16:
> Hi Astroxyx,
> We expect to have a version for 20.04 by the end of this week.

Thank you very much.

pabloinigoblasco, 2021-02-17:
We already have an initial version of the smacc_viewer in the noetic-devel branch of the smacc viewer.
Please, have a look to it.
Any feedback about errors or bugs is welcome.
Kind regards.

closed 2021-02-17 by pabloinigoblasco

astroxyx, 2021-02-18:
Hi Pabloinigoblasco,
But the smacc viewer is closed source,I can not install it in 20.04.

astroxyx, 2021-02-21:
Excuse me,can you tell me how can I install install smacc viewer in 20.04? It is closed source.

pabloinigoblasco, 2021-02-23:
Hello.

Have a look to these instructions: https://smacc.dev/smacc-viewer/
Please, replace the ROS version (from kinetic or melodic) to noetic.

Any feedback you may have is welcome.

Deeparaj-24, 2023-09-01:
I cannot get it to work in noetic

Belal111125, 2023-09-15:
apt-get is not working in noetic

Belal111125, 2023-09-15:
@Deeparaj-24  try to use the runtime analyzer=>  https://robosoft.ai/smacc1_rta-installation/