robosoft-ai/SMACC #27: Issue running SMACC atomic demo on melodic

zenified, 2021-05-25:
Hi I was following the getting started tutorial and faced issue running the atomic example. The following error
`smacc_runtime_test_node-3] process has died [pid 27462, exit code 255,` was printed after running `roslaunch sm_atomic sm_atomic.launch`

In addition, smacc_viewer_node, have both `wxversion 2.8 is not installed, installed versions are ['3.0-gtk3']` error message, and an exception with the following trace output
`Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/ros/melodic/lib/smacc_viewer/smacc_viewer_node.py", line 1914, in _update_server_list
    self.select_state_machine(state_machine_names[0])
  File "/opt/ros/melodic/lib/smacc_viewer/smacc_viewer_node.py", line 1963, in select_state_machine
    newtransitionhistory = self.transition_log_history_srv[server_name]()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
ServiceException: transport error completing service call: receive_once[/SmAtomic/smacc/transition_log_history]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
`
I wonder if there is any solution for this? Thanks

zenified, 2021-05-25:
Found that issue seems to be mainly on the smacc viewer.  The version provided by apt was 2.0.2. Pulling the latest copy from `https://github.com/ros-visualization/executive_smach_visualization` resolved the issue

closed 2021-05-25 by zenified

Wimll, 2021-07-26:
Thanks for the heads up, @zenified! I am having the same issues you are describing. Did the visualization of the current state also not work for you? 

Apt also installed a old version of smacc viewer for me (2.0.2). How exactly did you manually pull a newer version?

zenified, 2021-07-30:
Hi @Wimll, I pulled the source files from the github repository by running `git clone https://github.com/ros-visualization/executive_smach_visualization`, within the source directory of my catkin workspace.

You will need to build it after that, and that version should be called instead of the apt version when you call `rosrun`

Wimll, 2021-08-02:
> Hi @Wimll, I pulled the source files from the github repository by running `git clone https://github.com/ros-visualization/executive_smach_visualization`, within the source directory of my catkin [...]

Thanks @zenified ! However, if I pull this repo and build it the smacc_viewer command does not exist. Do I need to use a different command to run it?

zenified, 2021-08-03:
Hi @Wimll, did you source the workspace after building it? If you did, try
to see if you could find it using "rospack find", sometimes the ros tools
have issues resolving new packages, and doing that seems to help in
discovering new packages.

Wimll, 2021-08-03:
Hi @zenified, I indeed sourced the workspace after building. I get this error:

`rosrun smach_viewer smach_viewer.py 
Traceback (most recent call last):
  File "/home/willem/ros/noetic/system/src/executive_smach_visualization/smach_viewer/scripts/smach_viewer.py", line 36, in <module>
    from smach_msgs.msg import SmachContainerStatus,SmachContainerInitialStatusCmd,SmachContainerStructure
ModuleNotFoundError: No module named 'smach_msgs'`

The viewer complains about smach_msgs not being installed, but I want to use smacc_msgs instead since I only use smacc, not smach. Does this mean that I will have to change all the dependencies of the viewer?

zenified, 2021-08-03:
Hi @Wimll, terrible sorry, looks like I gotten myself confused there when I working on this. Smach and smacc are fundamentally different thing.  I am not sure why it worked, but somehow after building the Smach package menioned above, the smacc_viewer was able to work for me, and  I kinda got ahead myself there. 

Since both are python based, perhaps they shared a common lib.

Try running smacc_viewer again, if problem still persist  you might have to reopen this ticket. My apologies again.

pabloinigoblasco, 2021-08-09:
Indeed, the smach_viewer is one thing and smacc_viewer is another different tool.
Smach state machines does not work on smacc_viewer and smacc state machines does not work on smach_viewer.

Kind regards.