robosoft-ai/SMACC #28: GPG ERROR when installing smacc-viewer

ZhenshengLee, 2021-05-27:
## Environment

Ubuntu2004

## Behavior

When I execute cmd

```sh
curl -s  https://b0e12e65a4f16bfc4594206c69dce2a49a5eabd04efb7540:@packagecloud.io/install/repositories/reelrbtx/SMACC_viewer/script.deb.sh  | sudo bash
```

I ran into such error output.

```sh
W: GPG error: https://packagecloud.io/reelrbtx/SMACC_viewer/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F504394FF0062D60
E: The repository 'https://packagecloud.io/reelrbtx/SMACC_viewer/ubuntu focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Usage of apt_auth.conf(5) should be preferred over embedding login information directly in the sources.list(5) entry for 'https://packagecloud.io/reelrbtx/SMACC_viewer/ubuntu'

```

## Comment

1. How to solve this installation problem?
2. Is there aarch64 version of smacc-viewer?

Thanks!

ZhenshengLee, 2021-05-27:
@pabloinigoblasco

ZhenshengLee, 2021-05-28:
I found something on the website.

[image]

pabloinigoblasco, 2021-05-28:
Hello VictorLee.
It would be great to have smacc_viewer debian package for arm64 (I guess you are working in ROS and nvidia jetson)
and there is not any special reason why smacc_viewer couldnt work on arm64 because it is mostly python code.  

However that is a quite low priority task because in most of the cases developers work on a x64 machine. Assuming you are using the nvidia-jeston machine: I recommend you as a workaround to interact with it from other x86-workstation in the same ROS network so that you could use smacc_viewer there to see how the nvidia-jetson state machine behaves.

ZhenshengLee, 2021-05-29:
> Hello VictorLee.
> It would be great to have smacc_viewer debian package for arm64 (I guess you are working in ROS and nvidia jetson)
> and there is not any special reason why smacc_viewer couldnt work on arm64 (since it is mostly python code)
> 
> However that is a quite low priority task because in most of the cases developers work on a x64 machine. Assuming you are using the nvidia-jeston machine: I recommend you as a workaround to interact with it from other x86-workstation in the same ROS network so that you could use smacc_viewer there to see how the nvidia-jetson state machine behaves.

Thanks for your advice, and It's a good idea to work with remote ROS.

But what I really want to say is,
there is no smcc_viewer installation candidate in my x86 ubuntu20.04 PC due to the GPG error
when I followed instructions in [smcc.dev](https://smacc.dev/smacc-viewer/)

Thanks!

pabloinigoblasco, 2021-05-30:
Hello Victor.
Then I will have a look to that.
That approach should be available.

Thanks.

ZhenshengLee, 2021-06-03:
> Hello Victor.
> Then I will have a look to that.
> That approach should be available.
> 
> Thanks.

Visiting `https://packagecloud.io/reelrbtx/smacc_viewer` shows

[image]

pabloinigoblasco, 2021-06-09:
Hello Victor.
At the current moment the only debian package version for smacc viewer is amd64, there is not yet x86 and not arm nor arm64.
Those are in the list of pending tasks, but are less prioritary than other issues.

ZhenshengLee, 2021-06-15:
> Hello Victor.
> At the current moment the only debian package version for smacc viewer is amd64, there is not yet x86 and not arm nor arm64.
> Those are in the list of pending tasks, but are less prioritary than other issues.

But **the platform** I talked about **is** actually **amd64**, and that is a DELL PC.

closed 2021-08-01 by ZhenshengLee