robosoft-ai/SMACC #29: release via bloom

tylerjw, 2021-05-27:
I searched the ros index site and rosdistro repo and was not able to find smacc.  Would you mind doing a bloom release of this repo for melodic and noetic?

pabloinigoblasco, 2021-05-27:
I think that is a great feature.

We have that task on the table but it is not clear when it will be done (we are quite focused now in Smacc for ROS2)

Did you try the packagecloud debian file repository?
https://packagecloud.io/reelrbtx/smacc

Isn't that solution good enough for you?

tylerjw, 2021-05-27:
I didn't realize you have it via packagecloud... imo that's a better solution for me.  Thank you.

Timple, 2022-04-01:
Well, I could not find it either.

Could this be put back on the table for discussion? I did several bloom releases of other packages already. 
If we decide to go with smacc (as opposed to behavior-trees), would it be okay if I release it to the buildfarm?

The packagecloud option doesn't really work for us because we use `rosdep install`. And I'm not in favor of scripting around this installation for all our computers and robots.

destogl, 2022-04-04:
Hi @Timple,

It would be great to help us with the release of SMACC. Please let us know if you need anything, e.g., access rights or anything else.

Timple, 2022-04-04:
Alright, nice to know you're open to it. We'll continue evaluation :slightly_smiling_face:

closed 2022-11-24 by sync-by-unito[bot]