robosoft-ai/SMACC #33: Any plan to release arm64 deb packages?

ZhenshengLee, 2021-05-28:
Hi, 

First I want to say it's a great work for statemachines with ROS.

And I wonder if there is any plan to provide arm64 based deb packages so we can deploy it into arm based systems like NVIDIA Jetson devices.

pabloinigoblasco, 2021-05-28:
Hello VictorLee.
Thank you very much for the interest on the SMACC project.

I think that having ARM64 packages is a great feature. That is also on the table and should not be very hard to implement. However it is not a high priority task since we are quite focused in ROS2 and there are many things to do.

Nonetheless it would be great if somebody could help us in that task.
I think a good path to follow would be to modify the https://github.com/reelrbtx/SMACC/blob/noetic-devel/smacc_ci/packagecloud_docker/Dockerfile

so that we can create a basic cross-platform ros build environment. In order to do that it would be required to install arm build tools and also adding the arm64 ros repositories (in this case for noetic). 

At the end of that file there is some custom python script that would be a bit more complicated to update since the access to the package cloud system is not public (but we could handle that part if the previous one is done)

ZhenshengLee, 2021-05-29:
Thanks, and I will generate the packages with some scripts according to smcc_ci.

pabloinigoblasco, 2021-05-30:
That would be amazing!
Do not hesitate to ask if you have any blocker.
Thank you.

closed 2021-06-08 by ZhenshengLee