robosoft-ai/SMACC #41: SMACC viewer -> Current state not highlighted?

Wimll, 2021-07-27:
Hello!

I am using SMACC with the SMACC viewer on ros noetic. The sm_atomic demo runs fine, but the current state is not highlighted in the SMACC viewer on the left side of the viewer. Transitions are shown in the trnasition log however.

Any idea how I can fix this?

Thanks!

pabloinigoblasco, 2021-07-27:
I think that should be working. However, there might be some regression because of the latest code updates. I will have a look to that.

Could you please attach a picture of the issue you have?

Kind regards.

Wimll, 2021-07-28:
I am using the debian package of the smacc-viewer (apt get). The smacc viewer is not listed on github because it is closed source, right? Or how can I pull the latest version? I noticed the version provided by apt-get is 2.0.2.
This is the view I get:

[image]

The transitions are shown correctly on the right, but the current state is not indicated on the left. For complicated state machines this becomes a problem.

Furthemore, the smacc_runtime_test_node dies as I start the sm_atomic demo, could this be related to the issue?

[image]

Wimll, 2021-08-02:
@pabloinigoblasco Did you have any chance to look into this issue?

pabloinigoblasco, 2021-08-09:
Hello.
Is this issue related with https://github.com/reelrbtx/SMACC/issues/27?

Could you please paste the log file contents it is referenced in the error message?

deep-in-the-pc, 2021-08-11:
Hello.
I have also ran into the state and event transition not updating problem using the ros noetic version smacc-viewer. After looking into smacc_viewer_node.py, it seems there is a problem when using python 3 where the key to the self.widget.graph.subgraph_shapes dictionary is encoded but the itemname is not.

The quick fix I added was using ".encode('UTF-8')" to encode the itemname. This seemed to fix the problem with updating the current state colour.

```
         if oldmsg is not None:
            for state in oldmsg.current_states:
                if state in self.smacc_object_graphic_item_mapping:
                    itemname = self.smacc_object_graphic_item_mapping[state].encode('UTF-8')
                    if itemname in self.widget.graph.subgraph_shapes:
                        nodeitems = self.widget.graph.subgraph_shapes[itemname]
                        for node in nodeitems:
                            if "Bezier" in str(node):
                                # fillcolor
                                node.pen.color = (0.0, 0.0, 0.0, 1.0)
                                node.pen.fillcolor = (1.0, 1.0, 1.0, 1.0)
                                node.pen.linewidth = 1

        for state in msg.current_states:
            if state in self.smacc_object_graphic_item_mapping:
                itemname = self.smacc_object_graphic_item_mapping[state].encode('UTF-8')
                isleafnode = True
                stateinfo = [
                    stx for stx in self.state_machine_msg.states if stx.name == state]

                isleafnode = False
                if any(stateinfo):
                    isleafnode = not any(stateinfo[0].children_states)
                if itemname in self.widget.graph.subgraph_shapes:
                    nodeitems = self.widget.graph.subgraph_shapes[itemname]
                    for node in nodeitems:
                        if "Bezier" in str(node):
                            # fillcolor
                            node.pen.color = (0.0, 0.0, 0.0, 1.0)
                            if isleafnode:
                                node.pen.fillcolor = (
                                    255.0/255.0, 252.0/255.0, 164.0/255.0, 1.0)
                                node.pen.linewidth = 3
                            else:
                                #node.pen.fillcolor = (255.0/255.0,250.0/255.0, 77.0/255.0, 1.0)
                                #node.pen.fillcolor = (1.0, 1.0, 1.0, 1.0)
                                #node.pen.color = (200.0/255.0, 200.0/255.0, 47.0/255.0, 1.0)

                                node.pen.color = (0, 0, 0, 1.0)
                                node.pen.linewidth = 15

```
I haven't looked into the "Unknown source" in the events but I assume it is also a misbehavior from transitioning into python 3.

Regards.

Wimll, 2021-08-17:
Thanks @deep-in-the-pc , I will try your solution ! But since the source code is closed source, how do you change the smacc_viewer_node.py?

deep-in-the-pc, 2021-08-17:
The smacc_viewer_node.py should be located in /opt/ros/noetic/lib/smacc_viewer . Do keep in mind that if the package is updated, any changes to the files will be overwritten.

Wimll, 2021-08-19:
Thanks a lot @deep-in-the-pc ! This seems to be the exact problem I was facing and everything is running fine now!

closed 2021-08-19 by Wimll

pabloinigoblasco, 2022-02-03:
Noetic debian files for smacc_viewer are updated now in package-cloud with the proposed fix.
Thank you very much for your insights.