robosoft-ai/SMACC #46: How to use orthogonal lines?

Mr-Niels, 2021-11-08:
How can one use multiple parallel orthogonal lines in a SMACC state machine as illustrated below?

[image]

I tried using the examples from [here](https://github.com/pabloinigoblasco/SMACC/tree/master/examples/radial_motion_example) but they don't seem to work.

reelrobotics, 2021-11-09:
Hello Mr. Niels:)
Ok, so I think the first step is to not use the here link in your message, its way too old. Lets use master (or perhaps Noetic) going forward.

Ok, so for something like the example above, I would look to the sm_dance_bot series, and we'll talk about the best one which is sm_dance_bot_strikes_back....https://github.com/reelrbtx/SMACC/tree/master/smacc_sm_reference_library/sm_dance_bot_strikes_back

This example has multiple orthogonals, which you can see the code for here...
https://github.com/reelrbtx/SMACC/tree/master/smacc_sm_reference_library/sm_dance_bot_strikes_back/include/sm_dance_bot_strikes_back/orthogonals

And without knowing too much about your application (ToolSubstate?), I would recommend the Updatable Publisher -  https://github.com/reelrbtx/SMACC/blob/master/smacc_sm_reference_library/sm_dance_bot_strikes_back/include/sm_dance_bot_strikes_back/orthogonals/or_updatable_publisher.h

For how to orchestrate the different orthogonals, I would look at st_aquire_sensors  https://github.com/reelrbtx/SMACC/blob/master/smacc_sm_reference_library/sm_dance_bot_strikes_back/include/sm_dance_bot_strikes_back/states/st_acquire_sensors.h 

and navigate_to_waypoints_x https://github.com/reelrbtx/SMACC/blob/master/smacc_sm_reference_library/sm_dance_bot_strikes_back/include/sm_dance_bot_strikes_back/states/st_navigate_to_waypoints_x.h

To get started. Cheers.

Mr-Niels, 2021-11-10:
Hi @reelrobotics,
Thanks for your swift reply. It might've not been completely clear from my initial post, but what you suggest is not entirely what I'm aiming for. 

I am looking to set up a mode state that contains two parallel sub state machines (orthogonal lines) that run simultaneously side-by-side. That is (as far as I understand) exactly what is done in the here link, but also described in the repository's [readme](https://github.com/pabloinigoblasco/SMACC#orthogonal-lines).

If you look at [navigate_to_end_point.h](https://github.com/pabloinigoblasco/SMACC/blob/master/examples/radial_motion_example/include/states/navigate_to_end_point.h), you can see that this state actually has two initial child states, as it is inherited in the following fashion:
```
struct NavigateToEndPoint : SmaccState<NavigateToEndPoint, RadialMotionStateMachine,
                 mpl::list<NavigationOrthogonalLine,ToolOrthogonalLine>>
```
where the last argument is a list of the initial states `NavigationOrthogonalLine` and `ToolOrthogonalLine`.

I hope it is more clear now on what I am trying to achieve.

Is such a concept still possible at all in the current state of the smacc framework?

closed 2022-11-03 by brettpac