robosoft-ai/SMACC #48: [BUG] Linux output redirection causes segfault.

ZhenshengLee, 2021-12-28:
This odd_manager_node is a SMACC-based ros node that handles system error of robotics.

```sh
/home/nvidia/catkin_ws_drivers/devel/lib/odd_manager/odd_manager_node > /dev/null 2>&1
Segmentation fault
```

Any Idea of avoiding this?

Thanks.

pabloinigoblasco, 2022-02-03:
Hello.
I miss some information in your case.  Today we introduced some fixes, that may be related with your issue.
We try to reproduce your issue with a basic state machine and we do not get that error:

`
$ ./devel/lib/sm_atomic/sm_atomic_node > /dev/null 2>&1
`
Please, make us know if that issue persists.
Kind regards.

closed 2022-02-03 by pabloinigoblasco

ZhenshengLee, 2022-02-07:
> Please, make us know if that issue persists.

This issue has been fixed.

Thanks for all!