robosoft-ai/SMACC #49: SMACC Viewer does not show the currently active state

lucascoelhof, 2021-12-29:
I'm running SMACC on Ubuntu 20.04 using ROS noetic. I was trying to run the examples, and even though the state machine itself seems to be working fine, I cannot see the currently active state on the viewer.

[image]

I have installed SMACC Viewer using apt (```sudo apt-get install ros-noetic-smacc-viewer```) but I also had to install wx manually using ```sudo pip install -U wxPython```. Please let me know if I can assist with something else.

Mr-Niels, 2022-01-11:
I encountered the same issue, have a look at issue #41, there a solution is given. Specifically in this comment: https://github.com/reelrbtx/SMACC/issues/41#issuecomment-896988990

pabloinigoblasco, 2022-02-03:
Noetic debian files for smacc_viewer are updated now in package-cloud with the proposed fix.
in #41.
Thanks.

closed 2022-02-03 by pabloinigoblasco