robosoft-ai/SMACC #51: State machine crashes on state transitions

Mr-Niels, 2022-03-02:
This seems to be caused by state reactors and client behaviours that are destructed while their callback functions are running. 

Is there a way to prevent such scenarios? 
For custom objects it is possible to use mutexes to circumvent the issue, but for standard objects such as 'sr_all_events_go' this is less than ideal.

yassiezar, 2022-04-18:
We ran into a similar issue. We've had some success using a combination of mutex locks in the CBs' `onExit` methods to prevent callbacks from running during destruction and using SMACC's built-in [asynchronous CBs](https://github.com/reelrbtx/SMACC/blob/master/smacc/include/smacc/smacc_asynchronous_client_behavior.h) that works better with long-running `onExit` calls

yassiezar, 2022-05-05:
@pabloinigoblasco that's a fair point. I tried to make a toy example to see if I can replicate the behaviour, but was unable to since the signals you described seemed to be doing their job preventing the short-lived objects from being dereferenced too early. I'll need to think more on the problem, but I'll try to make a proper sample to recreate the issue for you.

For some additional context on my thinking: in our case, the crashes (always segfaults referencing non-existing objects) were non-deterministic and happened randomly, but always upon a user-triggered state transition and involved long-running CUDA image processing tasks. This is what made me suspect a race condition being somewhere and trying mutexes to prevent premature destruction of the associated CBs. This seems to have helped, but I agree its not a very elegant solution and doesn't quite fit into the SMACC style.

closed 2022-11-03 by brettpac