robosoft-ai/SMACC #52: Transition back to previous state using history

yassiezar, 2022-03-14:
Hi, I have a state, stateA, which is accessible from nearly every other state, stateB, stateC, etc. (similar to the recovery and operation ModeStates from the `SmDanceBot` examples). However, I'd like to implement a transition from stateA back to whichever state it transitioned in from - kind of like an interrupt state. For example,

stateB -> stateA -> stateB
stateB -> stateC -> stateA -> stateC

etc.

I've tried using the ModeStates from the DanceBot examples, but it looks like the transition back into the run modestate takes you to the pre-defined initial state, i.e. StAquireSensors, which isn't quite what I'm looking for. Is there some other way to do this? Can we possible leverage the `deep_history` stored by `Boost::StateChart`?

reelrobotics, 2022-03-14:
Hi Jaycee,
We do have an example using this functionality in the SMACC2 repo...
[https://github.com/robosoft-ai/SMACC2/tree/master/smacc2_sm_reference_library/sm_advanced_recovery_1](https://github.com/robosoft-ai/SMACC2/tree/master/smacc2_sm_reference_library/sm_advanced_recovery_1)

with the key line of code existing at line 31 in this file/state...
[https://github.com/robosoft-ai/SMACC2/blob/master/smacc2_sm_reference_library/sm_advanced_recovery_1/include/sm_advanced_recovery_1/states/ms_recover_inner_states/st_recover_step_4.hpp](https://github.com/robosoft-ai/SMACC2/blob/master/smacc2_sm_reference_library/sm_advanced_recovery_1/include/sm_advanced_recovery_1/states/ms_recover_inner_states/st_recover_step_4.hpp)

And here is a short video of sm_advanced_recovery_1 in action...
[https://vimeo.com/649655394/f9b25be7f9](https://vimeo.com/649655394/f9b25be7f9) at minute 8:40

There are also some snippets from around the SMACC1 library, but you may need to add an event or something to the example to get the specific transition to work...

[smacc_sm_reference_library/sm_ferrari/include/sm_ferrari/mode_states/ms_recover.h](https://github.com/reelrbtx/SMACC/blob/76dceb90411c2e4b13e4ae78c0af67d1acfb3333/smacc_sm_reference_library/sm_ferrari/include/sm_ferrari/mode_states/ms_recover.h)

[smacc_sm_reference_library/sm_three_some/include/sm_three_some/mode_states/ms_recover.h](https://github.com/reelrbtx/SMACC/blob/76dceb90411c2e4b13e4ae78c0af67d1acfb3333/smacc_sm_reference_library/sm_three_some/include/sm_three_some/mode_states/ms_recover.h)

But basically this should work for SMACC1 as well. Please keep us updated with how this goes:)

yassiezar, 2022-03-16:
Thanks for getting back to me so quickly.

I've gone through those examples and implemented them into my code. I'm happy to report that moving the non-interrupting states to a ModeState and accessing its history seems to send the interrupting state back to the correct state.

I think what threw me off was that the `LastDeepState` in the `smacc::SmaccState` class is typedef'ed to `InnerInitial`, which looked to me to be the initial state of the ModeState or StateMachine, depending on how it was defined (see [here](https://github.com/reelrbtx/SMACC/blob/76dceb90411c2e4b13e4ae78c0af67d1acfb3333/smacc/include/smacc/smacc_state_base.h#L34)). Also, it looked too easy haha. However, the SMACC2 example and video you posted cleared up my confusion.

One other question I had: what's the difference between the deep and shallow histories?