robosoft-ai/SMACC #56: Any support to smacc_action_server_client.h ?

leongkhing, 2022-05-24:
Hi, I am planning to do GUI that interacts with the state machine with action call. However, currently there is no action_server client available in SMACC, or am I missing something? Alternative would be to have another interface that receives the action call from the GUI and interacts with the state machine through service call. 

Any suggestion are welcome. Thanks!

pabloinigoblasco, 2022-07-22:
Hello.
There isn't any action server "smacc client" nor component to do that at the current moment.
But PRs on that are welcome.

pabloinigoblasco, 2022-12-29:
At the current moment in SMACC2 there is a stand-alone CbCallService client behavior.
It would be easy to backport from that file to SMACC1.