robosoft-ai/SMACC #65: Suppress logging messages from SMACC

sarahallen111, 2022-09-16:
I am working on a project that utilizes the SMACC library. We are using ROS logging for our package, but the logs are being spammed by the logging calls coming from inside the SMACC library. Is there any way to suppress the SMACC library logging messages, while still allowing the logging from our package?

logdan, 2022-09-19:
Hi, thanks for the response.

We can output the logs of the entire node to xterm, but there's currently nothing in place to let us separate the library logs from our SMACC implementation's logs, since they are the same node, correct?

pabloinigoblasco, 2022-09-19:
I understand your concern.

I am thinking now and I think that also should be possible just setting the smacc console level to WARN or to ERROR or to FATAL. This would make the logging system skip DEBUG an info messages.

Please have a look to these references:

https://github.com/reelrbtx/SMACC/blob/76dceb90411c2e4b13e4ae78c0af67d1acfb3333/smacc_sm_reference_library/sm_dance_bot/config/rosconsole.config

https://github.com/reelrbtx/SMACC/blob/76dceb90411c2e4b13e4ae78c0af67d1acfb3333/smacc_sm_reference_library/sm_dance_bot/launch/sm_dance_bot.launch#L22

We could use something like:

```
log4j.logger.smacc=FATAL
```
you might need to replicate this with other smacc client libraries, but I think it should work.

I think we should use this rosconsole file for all of our example, making easy to the user configure them.

Please, make us know if that works for you.

katet-tudelft, 2022-10-10:
I do something similar, with the rosconsole.config containing something like this:

```
log4j.logger.ros.smacc=ERROR
log4j.logger.ros.my_impl_name=INFO
```
This seems to suppress the smacc library messages, while allowing INFO from my_impl_name through.

brettpac, 2024-05-10:
Closing this one out as answered.

closed 2024-05-10 by brettpac