robosoft-ai/SMACC #97: apt update failed, 402 Payment Required

DeboBurro, 2023-04-17:
after running 
```
curl -s https://packagecloud.io/install/repositories/reelrbtx/SMACC/script.deb.sh | sudo bash
```
then I got error from apt update
```
Err:8 https://packagecloud.io/reelrbtx/SMACC/ubuntu focal InRelease
  402  Payment Required [IP: 54.183.84.206 443]
Reading package lists...
E: Failed to fetch https://packagecloud.io/reelrbtx/SMACC/ubuntu/dists/focal/InRelease  402  Payment Required [IP: 54.183.84.206 443]
E: The repository 'https://packagecloud.io/reelrbtx/SMACC/ubuntu focal InRelease' is not signed.
The command 'bash -c ./ros_dependencies.bash' returned a non-zero code: 100
```

pabloinigoblasco, 2023-04-17:
I wanted to bring a couple of things to your attention.

Firstly, if you are attempting to install SMACC or SMACC2 Debian packages, please note that they are no longer hosted on the Package Cloud. They are now officially maintained as ROS packages and can be found in the standard Noetic or Melodic repositories.

Secondly, if you are looking to install smacc_rta or smacc2_rta, please refer to the Robosoft webpage. The SMACC_RTA Debian packages are no longer available on the Reelrbtx Package Cloud. Instead, they can be found on the Robosoft Package Cloud.

https://robosoft.ai/smacc2_rta-installation/

DeboBurro, 2023-04-18:
I am able to apt install `ros-noetic-smacc` however some smacc owned packages like `ros-noetic-multirole_sensor_client`
  `ros-noetic-ros_timer_client` `ros-noetic-sr_all_events_go` `ros-noetic-ros_publisher_client` `ros-noetic-keyboard_client` don't seem to be officially maintained by ROS yet. or do you know where the apt repository url is to download these apt pacakges?

pabloinigoblasco, 2023-04-20:
Currently, only the core packages are being deployed in the ROS repositories that are actively maintained by us.

If you require clients and reactors for your project, you are expected to select and copy them directly to your workspace. This process has been working effectively.

However, we are in the process of selecting additional packages to be included in all ROS and ROS2 repositories, expanding the available resources for our users. So hopefully, they will be also available for you via apt very soon.

Thank you.

brettpac, 2024-05-10:
Closing this one as answered.

closed 2024-05-10 by brettpac