robosoft-ai/SMACC #118: Cannot install smacc_viewer in ubuntu 20.04

Deeparaj-24, 2023-09-01:
- I followed the instructions in [Link](https://smacc.dev/smacc-viewer/) , but I am not able to get it running. 
[image]
I am just getting the above error.

brettpac, 2023-09-01:
Hi @Deeparaj-24,
We retired the SMACC Viewer a couple of months ago. It just got too expensive to maintain. But we created a version of the new SMACC Run-Time Analyzer for or ROS1 users, and still free for individual and academic use. Here are the installation instructions... https://robosoft.ai/smacc1_rta-installation/

Please lemme know if you run into any issues. Thank you.

Deeparaj-24, 2023-09-02:
I installed the individual smacc_rta version but it is asking f
[image]
or the licence key.

brettpac, 2023-09-02:
Yup, so add the appropriate product to the cart and checkout... https://robosoft.ai/product-category/smacc1-runtime-analyzer/
and you'll get the license key you need.

Deeparaj-24, 2023-09-02:
Got it. Thank you so much

closed 2023-09-02 by brettpac